RobTillaart/I2CKeyPad #13: Hi

mdjavedakhtar, 2022-01-19:


RobTillaart, 2022-09-18:
@mdjavedakhtar 
Sorry for the late reply, apparently I missed some issues in the past.
I'm checking all my libs for missed issues, so can take a few days.

RobTillaart, 2022-09-18:
@mdjavedakhtar 

Some thoughts about your question. 
No, this library cannot support a 5x4 keypad as the I2C PCF8574 only has 8 pins.
You could however write a similar library based upon a PCF8575 that has 16 pins.

Adafruit has a [TCA](https://github.com/adafruit/Adafruit_TCA8418) which is a device meant for larger keypads.

Don't know if your question is still valid or solved otherwise,

RobTillaart, 2022-09-28:
@mdjavedakhtar 

Today I worked on extending the I2CKeyPad class, now it supports 4x4 5x3 6x2 and 8x1 keypads.
See the develop branch.

If time permits I will create a new class tonight **I2CKeyPad8x8** that supports an 8x8 matrix with the help of an PCF8575 IC.
So if all goes well there will be a develop version later this evening too, otherwise I'll continue to work on it tomorrow.

RobTillaart, 2022-09-29:
Released new 0.3.2 version of I2CKeypad class, to support 4x4 5x3 6x2 and 8x1 and smaller keypads.
So now focus on https://github.com/RobTillaart/I2CKeyPad8x8

RobTillaart, 2022-09-29:
@mdjavedakhtar 

https://github.com/RobTillaart/I2CKeyPad8x8 released. This library supports the 5x4 keypad.
If there are problems with it, please open an issue under https://github.com/RobTillaart/I2CKeyPad8x8

closed 2022-09-29 by RobTillaart