RobTillaart/I2CKeyPad #16: compilation error on arduino 2.04

GauchoMatrero, 2023-03-23:
Hi Rob,

I`m using your library along with a low profile oled ascii driver, and it worked ok till version 2.04 of arduino envionment.
I get a not in scope error for yield() function used in the loop that pools I2C to read the keyboard.
I did my research and found it's a weak function, for the user hook his own functions triggered each 7 us.
It's seldom used outside RTOs, but it has a weak declaration in arduino.h, so I added #include "arduino.h" in the I2CKeypad.h file, to solve the problem

RobTillaart, 2023-03-29:
Sorry for the late reply, I missed some notifications so I try to catch up.

>  it worked ok till version 2.04 of arduino envionment.

OK, newer than the version I use.

> I get a not in scope error for yield() function used in the loop that pools I2C to read the keyboard....
> ...so I added #include "arduino.h" in the I2CKeypad.h file, to solve the problem

Which version of the library do you use?
As I2CKeyPad.h already includes "Arduino.h" - https://github.com/RobTillaart/I2CKeyPad/blob/master/I2CKeyPad.h
I would expect it is not a problem.

Can you post your sketch (minimized to show the problem)

RobTillaart, 2023-03-29:
@GauchoMatrero 
What board do you use?

GauchoMatrero, 2023-03-29:
Mainly arduino duemillanove, for development purposes, and custom attiny for final product Enviado desde mi Galaxy
-------- Mensaje original --------De: Rob Tillaart ***@***.***> Fecha: 29/3/2023  12:22  (GMT-03:00) A: RobTillaart/I2CKeyPad ***@***.***> CC: Carlos Feldman ***@***.***>, Mention ***@***.***> Asunto: Re: [RobTillaart/I2CKeyPad] compilation error on arduino 2.04 (Issue #16) 
@GauchoMatrero
What board do you use?

GauchoMatrero, 2023-04-01:
This was the solution to my problem, I was modifying the library to use
TinyI2C instead of wire, and Added the include "arduino.h"  which solved my
compilation error.
Thanks for your help.

BTW. Studding this I learned I can use yield() to have a simple blinking
led, with no timer programming. This is my code.

/ yield task & auxiliary data for blinking
// Global data
volatile bool BlinkFlag{ false };
constexpr uint32_t TICKS_PER_SECOND = 142857;
uint32_t ticks{ TICKS_PER_SECOND };

void yield(void) {
  if (BlinkFlag) {
    if (--ticks == 0) {
      ticks = TICKS_PER_SECOND;
      digitalWrite(Locked_led, !digitalRead(Locked_led));
      digitalWrite(Pass_led, !digitalRead(Pass_led));
    }
  }
}

RobTillaart, 2023-04-01:
Ok, with that extra information (tinyI2C) I understand what was happening.
Good to hear you solved your problem.

Very creative use of the **yield()** function.👍
 It is not  what it was meant for but really out if the box !

If you want to you can share your modified library code (zip file). I could learn a bit about **tinyI2C** in practice.

GauchoMatrero, 2023-04-01:
Hi Rob,

I was trying to use smal OLED Displays along with a keyboard for my
projects, only with text.
The library  I used, allowed to use Tiny Wire, and it's really smaller so I
adapted your library to use Tiny I2C.
The libraries I used are:
#include "SSD1306Ascii.h"
#include "SSD1306AsciiWire.h"
#include "I2CKeyPad.h" ( modified by me )

And I'm attaching the modified library and my sketch. The other libs can
be  downloaded using the arduino environment.
Enjoy !

Carlos

RobTillaart, 2023-04-09:
As issue is silved I close the issue.

closed 2023-04-09 by RobTillaart