RobTillaart/I2CKeyPad #20: Problem with the numbers from keypad

victorMartins2024, 2024-06-10:
I'm developing a project that uses a password system to enter other operating screens, so I can't enter the password, when you open the first key it gives a password error filled in.

`void eng(){

  lcd.clear();
  lcd.setCursor(5, 1);
  lcd.print("PASSWORD:");
  lcd.setCursor(14, 3);
  lcd.print("#-SAIR");

  while (1) {

    char key = kpd.getChar();

    if (kpd.isPressed()) {
      vTaskDelay(20);
      if (key == 'C') {
        resetPassword();
      } else if (key == '#') {
        apx();
      } else if (key == 'D') {
        if (passvalue == true) {
          aprovadoPass();
          passvalue = false;
        }
      } else {
        processNumberKey(key);
      }
    }
  }
}

void processNumberKey(char key){

  lcd.setCursor(a, 2);
  lcd.print("*");
  a++;

  if (a == 8) {
    a = 4;  // Tamanho da senha com 4 digitos "2552"
  }

  currentpasslen++;
  password.append(key);

  if (currentpasslen == maxpasslen) {
    aprovadoPass();
  }
}`

RobTillaart, 2024-07-14:
Sorry for the long delay, I missed the notifications.

RobTillaart, 2024-07-14:
@victorMartins2024 
Is this problem still actual ? as it is 5 weeks old.

RobTillaart, 2024-07-18:
As there is no answer to my question, I close the issue.
Feel free to reopen it if the problem still exists.

closed 2024-07-18 by RobTillaart