RobTillaart/I2CKeyPad #21: I2CkeyPad and PCF8575 not working together

ennio64, 2024-06-10:
I2CKeyPad version=0.4.0 and PCF8575 version=0.2.3 not working together. 
On stm32f411 individually they work well but together they conflict. I tried to change the I2C bus for pcf8575 (I would like to use I2C3 SCL=PA8 and SDA=PB4) but I couldn't do it.
Thanks

RobTillaart, 2024-07-14:
Sorry for the long delay, I missed the notifications.

RobTillaart, 2024-07-14:
@ennio64 

The possible I2C addresses of the PCF8574 and the PCF8575 do overlap.
So you should not use the same (default) addresses for both. 

Have you tried to give the PCF8574 and the PCF8575 different addresses?

Another option is to use a PCF8574A for the keypad as this has another address range.

RobTillaart, 2024-07-14:
Will add a section to the readme.md file like this

### I2C addresses

This library uses a PCF8574 or a PCF8574A chip.
These devices are identical in behavior although there are two distinct address ranges.

|  Type      |  Address-range  |  Notes                    |
|:-----------|:---------------:|:-------------------------:|
|  PCF8574   |  0x20 to 0x27   |  same range as PCF8575 !  |
|  PCF8574A  |  0x38 to 0x3F   |

@ennio64 

Is the problem still actual?

RobTillaart, 2024-07-14:
Created develop branch with new readme.md

ennio64, 2024-07-15:
Not is a addresses conflict, when I tried both on the same I2C bus they had different addresses and conflicted. In my project I solved it by using two different I2C buses.

RobTillaart, 2024-07-15:
@ennio64 
I understand that using two I2C busses works as you separate the devices.

If I recreate the setup of an PCF8574 and PCF8575 on one bus I see two different I2C addresses.
(using different addresses by means of the A0, A1 and A2 pin, and pull up resistors on SDA / SCL of course).
So I do not understand how your conflict looks like? 
Did you see only one I2C address? did you get a lot of read failures? 
How did the conflict expose itself?

If you have time, can you please elaborate on this? If no time, no problem, then I just close the issue.