RobTillaart/I2CKeyPad #25: Not working in esp32 devkit v1

HARIS4820, 2024-08-26:
Error failed to communicate with keypad 
I connected to pin 25,26 and configured as same. I testing sample code in the library. Its not working

RobTillaart, 2024-08-26:
Thanks for reporting this issue.
As I am rather busy it might take time before I can investigate.

First checks:
If you scan the i2c bus, us the device visible?
Are pull up resistirs in place for the SDA SCL lines?

RobTillaart, 2024-08-26:
Is the I2C bus configured to use pin 25 and 26?

RobTillaart, 2024-08-26:
[image]

(image from - https://mischianti.org/esp32-pinout-specs-and-arduino-ide-configuration-part-1/)

I2C is pin 21 and 22

HARIS4820, 2024-08-26:
I am using mfrc522 rfid so 21 22 already in use. Recommend pin 5v side. 

Thank you for your quick response

RobTillaart, 2024-08-26:
@HARIS4820 

But you can put multiple devices on one I2C bus as they all have (assumption) a different address.

You might need to use something like Wire1 and set

```cpp
Wire1.begin(25,26)
```

HARIS4820, 2024-08-26:
i made small change in sample code 

Wire1.begin(26, 25);  //  adjust pins if needed
  Wire1.setClock(400000);
  keyPad.begin();
  while (keyPad.begin() != true) {
    delay(1000);
    Serial.println("Attempting to communicate with keypad");
    WebSerial.println("Attempting to communicate with keypad");
  }

i connected i2c and keypad. and i also reboot multiple times. i am using i2c with lcd also. i am getting result
[image]

HARIS4820, 2024-08-27:
I a

HARIS4820, 2024-08-27:
I am using this i2c.
https://robu.in/product/pcf8574t-i2c-i-o-extension-board/

HARIS4820, 2024-08-27:
I am using this i2c https://robu.in/product/pcf8574t-i2c-i-o-extension-board/

RobTillaart, 2024-08-27:
What is the address set by this pcf8574 module?
Do you use that in your code?

HARIS4820, 2024-08-27:
const uint8_t KEYPAD_ADDRESS = 0x38;

I am using this in sample code

RobTillaart, 2024-08-27:
Does that address 0x38 match the address set by the jumpers on the board?
Do you call the constructor with the **KEYPAD_ADDRESS** and the **Wire1** I2C bus parameters?

```cpp
const uint8_t KEYPAD_ADDRESS = 0x38;

I2CKeyPad(KEYPAD_ADDRESS , &Wire1);
```

HARIS4820, 2024-08-27:
I call keypad address and wire1. How to find 0x38 matching or not?

RobTillaart, 2024-08-27:
> I call keypad address and wire1. How to find 0x38 matching or not?

You need to use an I2C scanner.
Connect only the PCF8574 to the default Wire bus (21, 22) and connect no other devices.
Run the I2C scanner 

Check my library - https://github.com/RobTillaart/I2C_SCANNER - in the examples.

HARIS4820, 2024-08-27:
I find bus address on 32. Thank you so much for your support

HARIS4820, 2024-08-27:
I tried scanner minimal sample code.
I change my code
const uint8_t KEYPAD_ADDRESS = 0x38;

To

const uint8_t KEYPAD_ADDRESS = 0x32;

But still facing same issue

HARIS4820, 2024-08-27:
[image]

RobTillaart, 2024-08-27:
32 is the decimal value
Use 0x20

HARIS4820, 2024-08-29:
I changed the code to 
const uint8_t KEYPAD_ADDRESS = 0x20;
And still facing same issue.

Thank you for your continuous support.

RobTillaart, 2024-08-29:
Can you try the example **I2Ckeypad_Wire1_ESP32.ino** and post the output?
- change address to 0x20  (line 16)
- change pins as needed (line 30)

HARIS4820, 2024-08-30:
Now working perfectly. 
Thank you 
Thank you so much for your guidance.

closed 2024-08-30 by HARIS4820