RobTillaart/I2CKeyPad #26: i update esp32

kharaziamir, 2025-04-15:
hello 
i update espressif/arduino-esp32 to v3.2.0 and this code is not work.
but with espressif/arduino-esp32 v 2.0.17 this work is correct.

please help me

#include "Wire.h"
#include "I2CKeyPad.h"
#include <LiquidCrystal_I2C.h>

const uint8_t KEYPAD_ADDRESS = 0x38;
static uint8_t lastKey = 'D';
static uint8_t keyup = 1;

I2CKeyPad keyPad(KEYPAD_ADDRESS, &Wire1);

uint32_t start, stop;
uint32_t lastKeyPressed = 0;
uint32_t value = 0;

LiquidCrystal_I2C lcd(0x27, 20, 4);  // set the LCD address to 0x27 for a 16 chars and 2 line display

void setup() {
  Serial.begin(115200);
  Serial.println(__FILE__);

  Wire1.begin(21, 22);  //  adjust pins if needed
  //Wire1.setClock(400000);
  if (keyPad.begin() == false) {
    Serial.println("\nERROR: cannot communicate to keypad.\nPlease reboot.\n");
    while (1)
      ;
  }
    lcd.init();  // initialize the lcd
    lcd.backlight();
    lcd.setCursor(0, 0);
    lcd.print("Start");

}

void loop() {
  uint32_t now = millis();
  if (now - lastKeyPressed >= 100) {
    lastKeyPressed = now;

    start = micros();
    char c = handleKeyPadValue(value);
    if (value > 125000) value = 125000;  //  some sample max.
    stop = micros();
    Serial.print(millis());
    Serial.print("\t");
    Serial.print(value);
    Serial.print("\t");
    Serial.print((char)c);
    Serial.print("\t");
    Serial.println(stop - start);
  }
}

//  handleKeyPadValue is used to read a uint32_t from the keypad
//  every digit must be pressed separately
//  the last key pressed is also returned.
//
//  0..9 adds to the number
//  * means remove last digit
//  # resets to zero
//  ABCD are not mapped.
char handleKeyPadValue(uint32_t &value) {
  char v[19] = "123A456B789C*0#DNF";  //  N = NoKey, F = Fail
  static uint8_t lastKey = 0;

  uint8_t idx = keyPad.getKey();
  char c = v[idx];

  if (lastKey != c) {
    lastKey = c;
    switch (c) {
      case '0' ... '9':
        value *= 10;
        value += c - '0';
        break;
      case '*':
        if (value > 0) value /= 10;
        break;
      case '#':
        value = 0;
        break;
      case 'A' ... 'D':
        //  e.g. store value in EEPROM
        break;
      case 'F':
        Serial.println("FAIL");
        break;
      case 'N':
        Serial.println("NOKEY");
        break;
      default:
        break;
    }
  }
  return c;
}

//  -- END OF FILE --

RobTillaart, 2025-04-15:
Had a quick look at it and the problem is NOT in my libraries.

It has been reported on the Arduino forum and Espressif forum a number of times.
See - https://forum.arduino.cc/t/esp32-3-2-0-i2c-fails/1373740

My advice is to revert to ESP32 - version 3.1.3 as that seems to be the last stable version (for I2C).

Rob

RobTillaart, 2025-04-17:
@kharaziamir 
As this problem is not a library problem I close the issue. 
Feel free to add remarks / progress / solutions if this issue is closed.

closed 2025-04-17 by RobTillaart